Meulengracht/MollenOS #58: [SYST] Thread Idle - 0 (Core 0)!   VM

Xoooo, 2016-09-08:
`/* Installs Timers */
void InitTimers(void)
{
	/* Setup Timers */
	DevicesInitTimers();

	/* Debug */
	LogInformation("TEST", "DevicesInitTimers()  End");
	
	/* Init Apic Timers */
	ApicTimerInit();

	/* Debug */
	LogInformation("TEST", "ApicTimerInit()      End");
	
	//After the above code is completed, Core 0 idle thread exception occurs
	
	for (;;);
	
	/* Boot Cores */
	CpuSmpInit();

	/* Debug */
	LogInformation("TEST", "CpuSmpInit()         End");
}`
The link below is a screenshot of an exception:
http://i4.buimg.com/1949/73cf625360d66815.png

Meulengracht, 2016-09-08:
Thanks again for reporting this issue, from the screen dump it seems like a missing null check, I'll take a look at it later, thank you for including the screen shot

Meulengracht, 2016-09-08:
This issue should be fixed now, see commit 329 for details

closed 2016-09-08 by Meulengracht