Meulengracht/MollenOS #170: Screenshot disappeared from repo

jsmaniac, 2018-10-15:
Hi!

You posted a cool screenshot of MollenOS's app search interface at https://forum.osdev.org/viewtopic.php?p=285451#p285451 a while back, but the link that you put points to the master branch and that file has since been deleted.

For archival purposes, could you put the screenshot back in `docs/images/appsearch.png`, or better change the link on the OSdev forum so that it points to a specific commit that contains the file (https://raw.githubusercontent.com/Meulengracht/MollenOS/2a71b8e2c44c737199a8244b11bb7ddd92ea61fa/docs/images/appsearch.png seems to be the commit that introduced it). Assuming you don't need to run `git filter-branch` in the future, the reference should not depend on changes in master that way (otherwise maybe make an `osdev-screenshot-appsearch-2018` tag or branch?).

Thanks a lot, and keep up the cool UI work :D !

Meulengracht, 2018-10-17:
hi jsmaniac,

thanks for your comment and sorry for the slow reply! I'll get the screenshot back for you!

Meulengracht, 2018-10-17:
otherwise with the new layout its actually just https://github.com/Meulengracht/MollenOS/blob/master/docs/images/gfx2.png

Meulengracht, 2018-10-17:
I updated the link to point to the old file on osdev! Have a great day

closed 2018-10-17 by Meulengracht