Meulengracht/MollenOS #171: Leaks

mrmajik45, 2018-10-25:
tools/bdf2c/bdf2c.c 

- [ ] Resource leak: _in

tools/lzss/main.c

- [ ] Resource leak: oldfile

tools/rd/main.c

- [ ] Resource leak: out

librt/libc/locale/lctype.c

- [ ] Memory leak: ctp

librt/libc/locale/lmessages.c

- [ ] Memory leak: mep

librt/libc/locale/lmonetary.c

- [ ] Memory leak: mop

librt/libc/locale/lnumeric.c

- [ ] Memory leak: nmp

librt/libc/locale/timelocal.c

- [ ] Memory leak: tip

librt/libc/stdio/fread.c

- [ ] Memory leak: readbuf

librt/libc/stdlib/malloc.c

- [ ] Memory leak: mem (1)
- [ ] Memory leak: mem (2) 

modules/serial/usb/xhci/main.c

- [ ] Memory leak: Transaction

tools/file2c/main.c

- [ ] Memory leak: Buffer

tools/lzss/main.c

- [ ] Memory leak: data

tools/rd/main.c

- [ ] Memory leak: copy
- [ ] Memory leak: tokens

userspace/zlib/gzlib.c

- [ ] Memory leak: path

Meulengracht, 2018-10-26:
Ill take a look at these asap, thanks for posting these.

mrmajik45, 2018-10-26:
No prob

mrmajik45, 2018-10-29:
Woo fixes! ;p

Meulengracht, 2018-10-29:
Haha - they are getting fixed in the proc-upd branch which i'm currently active in, i'll merge into to master when my work there is done. Also the memory leak in the XHCI file is just a placeholder file to keep the directory.

mrmajik45, 2018-10-29:
Well you need to remove Resource and memory leaks to keep a system from crashing

mrmajik45, 2018-10-30:
Where are the icons?

mrmajik45, 2018-10-30:
Might use imgbot. It auto-losslessly compresses images so they're smol but still good soo.