Meulengracht/MollenOS #211: Libcxx is not compiling.

Manish-Kala-Jaat, 2019-04-27:
When compiling the millennium os with the command :- make VALI_ARCH=amd64 CROSS=pointing toolchain.
It runs good till the Libyan.lib but after that the turn of getting compiling of libcxx comes but
It is giving the error saying that the := LIBCXX_ENABLE_STATIC_ABI_LIBRARY  requires Python  but it is not found.
Let me clear you this issue looks related to cmake but I have installed all the pre-requisites given in the makefile of mollen os and vali toolchain aiso.

 Here are the screenshots...

**SCREENSHOTS**

[image]
[image]

[image]

Meulengracht, 2019-04-27:
You must run the tools/depends.sh, it will install the Vali.cmake platform file required for using cmake build with vali

Meulengracht, 2019-04-27:
See link https://github.com/Meulengracht/MollenOS#setting-up-for-os-development

Meulengracht, 2019-04-27:
Also if you are building the 64 bit build please pull latest changes as they fixed an issue with VirtualBox startup

Meulengracht, 2019-04-27:
Please close this issue if you are able to compile after these instructions

Manish-Kala-Jaat, 2019-04-28:
Thanks man it worked .

closed 2019-04-28 by Manish-Kala-Jaat