Meulengracht/MollenOS #213: stuck at startup debug console in virtual box.

Manish-Kala-Jaat, 2019-04-29:
I just compiled the latest commit of mollen os repository and generated the vmdk file and tried to run it in the virtual box but it shows the startup debug console and stuck it out there.
the last line which it shows on the debug console is =
[LIBC] Spawning message handler.

**Screenshots**
[image]

Meulengracht, 2019-04-29:
Are you running with multiple cores? Just for my own information when I reproduce this later

Manish-Kala-Jaat, 2019-04-29:
check it up in this pic=

[image]

Manish-Kala-Jaat, 2019-04-29:
well for your help ,I am sending more screenshots=

[image]

[image]

[image]

Meulengracht, 2019-04-29:
Perfect thanks - I’ll look into it. It looks like the renderer is stuck. It most likely a side effect from the recent multi core improvements

Manish-Kala-Jaat, 2019-04-29:
sure

Meulengracht, 2019-04-29:
This is because of #198 . It needs to be fixed before the 64 bit version of llvm's code generation for shaders is working. I would recommend you use the 32 bit version of everything untill #198 is fixed.

Meulengracht, 2019-04-29:
You can see the error messages if you go into vali-userspace/vioarr/vioarr.cpp and comment out line 65

Manish-Kala-Jaat, 2019-04-30:
Sure man

Meulengracht, 2019-04-30:
This is due to something else. Discovered that issue #198 was not an issue after all :-) I'll keep looking into this!

Meulengracht, 2019-05-01:
I was using the wrong calling convention for the assembly function responsible for copying the back buffer to the screen. This resulted in some undefined behavior which most likely is the cause. If you get the latest source for vali-userspace this should be fixed now

Manish-Kala-Jaat, 2019-05-01:
I tried it but now it stuck in the ahead part , it shows the green screen .
screenshot: ->
[image]

Meulengracht, 2019-05-01:
Yes I discovered - I have yet to determine what the issue is on this issue, I have to investigate the difference between the i368 and amd64 build for Mesa. I do suspect a few things tho.

Meulengracht, 2019-05-02:
It'll take a few days for me to fix this most likely, or atleast understand what the issue is. (I don't have time before the weekend), untill then use the 32 bit distribution if you just want to try it.

Meulengracht, 2019-05-05:
The render issue has been resolved. It had to do with the wrong 64 bit calling convention used by mesa in its code generation (mollenos is using the same as windows, which needs to be specified). There is an issue with input (key events) in 64 bit mode. I'm looking into that. I'll open a new issue for that

closed 2019-05-05 by Meulengracht