Meulengracht/MollenOS #231: could MollenOS fit on a floppy?

informer2016, 2020-05-03:
Even today the floppies are still being used, for example - as virtual floppies inside the coreboot open source BIOS. Just imagine: your wonderful OS could be a part of someone's BIOS build! _(for coreboot supported motherboard, maybe you have or could get one - see https://www.coreboot.org/Supported_Motherboards )_

@Meulengracht , If you already have a coreboot-supported motherboard, or a real chance to get one, - wouldn't it be cool to be able to launch your own OS straight from the BIOS chip? ;) With one simple command its possible to add any floppy to coreboot BIOS build - and then you see it as a boot entry! Multiple floppies could be added this way _(as long as you have enough space left inside the BIOS flash chip, luckily LZMA compression could be used for the stored floppies to reduce their occupied size)_

Meulengracht, 2020-05-03:
Yes the boot loader, initial ramdisk and kernel could fit on a floppy disk (would require only 800-900kb) - however the user space would not be able fit, and would require external storage in order to use any graphical UI

informer2016, 2020-05-03:
@Meulengracht 2.88MB "double sized" floppy images are also supported by coreboot+SeaBIOS as well as QEMU and VirtualBox, although they are rare like unicorns in a physical world...

Meulengracht, 2020-05-03:
You would be able to run a console environment in my OS on less than 1,4mb. 

But you would need atleast 1gb for everything including the Windows manager and sol libraries

informer2016, 2020-05-03:
@Meulengracht @Meulengracht I see, indeed it may be hard to use the remaining 2MB for something truly useful, that's if your OS isn't written on ASM or C-- like KolibriOS. Thank you for reply, and have a nice day ;-)

closed 2020-05-03 by Meulengracht