Meulengracht/MollenOS #263: unable to build vali-userspace

mnisk, 2021-09-20:
I am getting this error...

mak@mak-VirtualBox:~/Downloads/mollen/App/vali-userspace$ cmake .
-- Install directory: /home/mak/Downloads/mollen/App/vali-userspace/install
-- GLM_INCLUDE_DIR = /home/mak/Downloads/mollen/App/vali-userspace/glm
-- Found GRLM in /home/mak/Downloads/mollen/App/vali-userspace/glm
-- Found GLFW3 in /usr/include
-- Using the builtin libz library sources
-- Using the builtin lzma library sources
-- Using the builtin bzip2 library sources
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find VIOARR (missing: VIOARR_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  SDL/cmake/FindVIOARR.cmake:50 (find_package_handle_standard_args)
  SDL/CMakeLists.txt:151 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/mak/Downloads/mollen/App/vali-userspace/CMakeFiles/CMakeOutput.log".

Meulengracht, 2021-09-20:
Hi mnisk, thanks for trying out building the userspace! Currently I'm in the process of updating my toolchain (mostly done) and userspace runtime libraries, getting c++20 and c++ filesystem to work - so the userspace is currently unstable and thus probably won't build correctly as I'm still making cmake changes to adapt to the updated llvm. They switched to a monorepo and I had to port all my changes to a new fork which has been very time consuming.

The postive things about this major redo is that I'm getting clang and lld ported as well now, as well as getting the entire toolchain updated to version 14, and C++20 support, including native toolchain and build formats. This makes it so much easier to build new applications for my OS!

I expect having this done within a week or two - as I'm currently finishing up the c++17 filesystem support, and implementing missing functionality (primarly links).

I'm very sorry for this inconvience, but I promise it'll be worth it for the OS, as we are getting a much nicer user experience and much prettier windowing system!

Meulengracht, 2021-09-30:
The configuration issues should now be solved. The CMakeSystem has changed to valicc, which means you need to run ./tools/depends script located in tools folder in this repository.

With that said, it does not build correctly yet. First of all you must rebuild the toolchain since a lot of changes has been done to llvm/clang port for Vali, and second of all I'm still implementing some missing support for my binary format so the userspace won't run yet. I'm keeping this issue open to write status updates for when everything should be working